bdring/midTbot_esp32 #1: Absolutely PHENOMENAL project!

MHeyster, 2019-08-02:
Just wanted to drop a message and say that this is an incredible project, it is extremely well implemented. I'd love to collaborate on the hardware side of you'd be up for it? Send me a message!

bdring, 2019-08-05:
What did you have in mind? We can discuss here or via my [Slack](https://join.slack.com/t/buildlog/shared_invite/enQtNjEzNjMwMTA3ODQ2LTZkMTE5MjcwZTY4MjgwOWEwMWE1MzZlNGNhNDdmMWEwNWZkN2Y5NjBlNjBlZjY2MTg3ZThlN2I3YjEwMzUyMWI)

BTW: I am on vacation in Colorado and will be off grid a lot.

jjgerber01, 2019-08-13:
Hi, 

I went ahead and had the PCB made by PCBGOGO in China. They came out great. Obviously they don't just fabricate 1 board, so I ordered the minimum of 5. I want to build one of these each for me and my daughter and keep a spare. Can I donate the other two boards to you or anyone you know deserving of this? 
Attached are some pictures. I can mail them out. 
Now, just to collect the other parts for it. 
Thanks for the contribution. 

J.J. Gerber 

43 Hollyberry Drive

Paradise, NL, A1L 4A1

Canada

Phone: +1 (709) 749-1612 
E-mail: jjgerber01@yahoo.ca

jjgerber01, 2019-08-13:
J.J. Gerber 

43 Hollyberry Drive

Paradise, NL, A1L 4A1

Canada

Phone: +1 (709) 749-1612 
E-mail: jjgerber01@yahoo.ca

PaulHaeberli, 2019-08-13:
Have you found a source for

https://www.digikey.com/short/p993q4

All the suppliers I've tried are out of stock, and it looks like they only

plan to get parts around Dec 26, 2019.

Also, is there a photograph of the board I can use as reference to figure

out where the parts are supposed to go?

Mfg Part Number: D2FS-FL-N-A
Customer PO Number:
Backorder Qty: 10
Ship Method: FEDX GROUND
*Original Delivery Date** *Qty* *Revised Delivery Date** *Qty*
*12/4/2019* 10 *12/26/2019* 10

jjgerber01, 2019-08-13:
I had the same problem, but this one matches mechanically. I hope it matches electrically too. You will have to solder some stiff wires inot the PCB and then attach the pins of this switch. The bolt holes aligh well. I inlcuded some pictures with a ruler frm the PCB I had made. 

Gikfun Micro Switch Long Hinge Lever (Pack of 20pcs) forArduino EK1713

by Gikfun_Official_Store

Learn more:https://www.amazon.ca/dp/B06XMQG4TC/ref=cm_sw_em_r_mt_dp_U_pKUuDbDW6YA32

I hope this helps

J.J. Gerber 

43 Hollyberry Drive

Paradise, NL, A1L 4A1

Canada

Phone: +1 (709) 749-1612 
E-mail: jjgerber01@yahoo.ca

 

https://www.digikey.com/short/p993q4

bdring, 2019-08-13:
I started a [new issue](https://github.com/bdring/midTbot_esp32/issues/2) to discuss sourcing the switch.

@jjgerber01

renschler, 2019-08-13:
@jjgerber01 if you still have PCBs to spare i'd be interested in buying one or two off you

@bdring great project! thanks for sharing