bdring/midTbot_esp32 #2: Limit Switch Sourcing

bdring, 2019-08-13:
The [switch on the BOM](https://www.digikey.com/short/p993q4) is hard to get.  Here are some possible alternatives.

[This version](https://www.digikey.com/product-detail/en/omron-electronics-inc-emc-div/D2FS-FL-N-A1/SW1985-ND/4753391) has a reversed lever. It will definitely work on the Y axis. It will probably work on the X, but the actuator post is not in the ideal location on the lever. 

[image]

[This amazon switch](https://www.amazon.ca/dp/B06XMQG4TC/ref=cm_sw_em_r_mt_dp_U_pKUuDbDW6YA32) won't work. You need right angle terminals.

[image]

This type of [switch from AliExpress](https://www.aliexpress.com/item/859187710.html) is fits. It just has the lever reversed like mentioned above.

Search for [DF2-L-A](https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20190813091359&origin=y&SearchText=D2F-L-A+&switch_new_app=y) on AliExpress for more options

[image]

dirkx, 2019-08-17:
[Farnell #2402447, €0.88](https://nl.farnell.com/omron-electronic-components/d2fs-fl-n-a/microswitch-lever-spst-0-1a-6vdc/dp/2402447) 
[mouser[(https://nl.mouser.com/ProductDetail/Omron-Electronics/D2FS-FL-N-A) and
https://nl.aliexpress.com/item/33032594245.html
https://nl.aliexpress.com/item/32802131724.html
https://nl.aliexpress.com/item/32881389932.html

bdring, 2019-08-18:
The design has been [updated](https://github.com/bdring/midTbot_esp32/commit/2eb69e2da447061a77404fdc40bd8de124e4595f) to allow both lever positions to work.

- The X actuator post was moved back 5.4mm
- The cover was modified to clear the lever.

closed 2019-08-18 by bdring

bdring, 2019-08-28:
I was wrong about the alternate lever position. The lever is not reversed, the leads are just bent the other way. This means it will switch around some pins if you solder it that way. It is basically upside down. This won't work right.

The simple fix is to just bend the leads back to the original position. I have done this on dozens of switches now and it is safe and effective.