bdring/midTbot_esp32 #3: Idler sizing

dirkx, 2019-08-14:
The BOM should probably stress that the idlers need to be fairly small; with an OD of the rims of 13 or even less. (as standard idlers for 2GT tend to be around 22-24mm).
[image]

bdring, 2019-08-14:
I typically use toothless idlers. Several of the idlers touch the back of the belt. 

They are usually matched to the equivalent tooth size. The ones I use are 16 tooth equivalent diameter.

I'll try to find  a way to link to link the BOM item to a page with more detailed dimensions.

bdring, 2019-08-14:
Here are the dimensions for the smooth idler.

[image]

dirkx, 2019-08-15:
On 15 Aug 2019, at 00:18, bdring <notifications@github.com> wrote:
> I typically use toothless idlers. 
> 

Drawing was a bit confusing. The main point is that most default idlers and pulls with tooth have a larger rim OD than the ones that currently easily fit. And this is rarely part of the spec-sheet.

So it would be good to update the BOM to stress the much smaller than `default' size of the outer rims. And ultimately adjust the design a bit perhaps to allow the more common 22-25mm outer diameter idles (the inner diameter is only 2-3mm larger than yours).

Dw

bdring, 2019-08-15:
I added more info to [BOM](https://github.com/bdring/midTbot_esp32/blob/master/Docs/mechanical_BOM.md) and a link to this issue.  Changing the pulleys would affect many parts, increase the size and reduce the work area. I don't want to do that.

closed 2019-08-18 by bdring