bdring/midTbot_esp32 #4: Width of the ESP32 Dev kit (0.96)

dirkx, 2019-08-14:
It appears that a fair number of ESP32 dev kits (13/26 pin version) sold as 0.96 wide are wider than what fits on the circuit board. Simply adding a row of 13 holes next to the current one; with the usual spacing, would accommodate both.

bdring, 2019-08-14:
That is a feature I have added to many of my other ESP32 projects. Typically it is a 19 pin connectors with 0.9" and 1.0" pitch.

Do you have details on the 0.96" part?

[image]

dirkx, 2019-08-15:
Yes - that layout should work splendidly - and there is just enough room for it.
I'll try to make a list of the amazon/aliexpress URLs with the 'sizes' .

dirkx, 2019-08-17:
So actual width on aliexpress, amazon & ebay for the units (17# units, 5 vendors) sold as 13 pin / 0.96"(24.4mm)  seems to be closer to 26mm if they are not exactly  0.96" as specced (3#). And 5 units are exactly 0.1" wider.  

So the extra row straight against the current 0.96 would cover all these nicely.

bdring, 2019-08-17:
You you have a link and picture for the 13 pin 0.96" modules?

dirkx, 2019-08-29:
Photo's and solution below.

[image]
[image]

bdring, 2019-08-29:
I am still unclear on the 13 pins?

JamesCWhite, 2019-08-30:
Pretty sure the .96" measurement from most listings refers to the oled size, even if it's a model without one. The way those sellers copy each other's listings, it's probably reused from a listing for one with a screen.

dirkx, 2019-08-30:
If - that -  I just got a batch from one DE reseller with 2 ‘narrow ones’ and 8 wider ones (so 0.96 and the other 2.54mm wider). And had the same a while ago on Aliexpress.

Guess the standard is ‘both’. :)