bdring/midTbot_esp32 #6: Feature/improvement: through the hole capacitors

dirkx, 2019-08-14:
Would be nice to also have hole-pads for the 4 SMD capacitors. So you have the option.

Probably not practical for the SD slot - as those are (too) hard to get in through the hole versions.

bdring, 2019-08-14:
These days most people are getting pretty comfortable hand soldering things like 1206 and 0805 caps. I don't want to change those parts.

The SD card is a PITA, but there are not too many options. On my first test controller, I used a standard SD card module. I don't know if there is a way to fit that and still make the card accessible.

[image]

bdring, 2019-08-18:
Checked: There is no good location for the standard micro SD card module.

[image]

closing issue

closed 2019-08-18 by bdring