bdring/midTbot_esp32 #8: Improvement request: screw terminal

dirkx, 2019-08-14:
Would be nice to have two screw terminal holes in addition to the normal barrel style socket. With +12v/GND labeling. For balance reasons - having it in the middle back of the board at the bottom may be nice.

bdring, 2019-08-14:
I don't want to increase the board size at all. The back concerns me. It could tend to rub on the Y axis and there is not much space, especially if a second row of CPU pins is added.

Here is a possible overlapping footprint, that allows you to chose barrel or terminal block.

[image]

dirkx, 2019-08-15:
Splendid - that would work. Would be nice to also label the silk screen.

dirkx, 2019-08-17:
Just rechecked, tried and find that a screw terminal just next to the bolt hole for the casing  and right of the 'V1" label at the back of the board works well - and 'balance' wise is nicer than the current barrel  connector location.

bdring, 2019-08-18:
I don't think it will fit after the dual footprint is added. 

Also keep in mind the cover needs over a 1mm clearance around the perimeter.