bdring/midTbot_esp32 #10: Update BOM / ease procuring of stepper motors

dirkx, 2019-08-17:
The currently specced/linked stepper motor link has a fairly short axis (12.5mm length).

Actual round axis length can be up to 22mm to still clear the left/right feet. 

So would suggest updating the BOM to allow up to 20.8mm of shaft length. And perhaps clarify that the mentioned 20mm body/enclosure size is not critical.

As that means there are many more stepper motor suppliers suddenly.

bdring, 2019-08-18:
The stepper motor must be a NEMA14 with 5mm shaft. 

The height of the body of the motor must be 20mm or less to fit the standard cover. A taller motor is fine if you don't use a cover or design a deeper one.

The motor shaft on the one listed in the BOM is rather short. I can be up to about 20mm long.

I have linked this issue to the BOM.

Closing issue.

closed 2019-08-18 by bdring

bdring, 2019-10-15:
[Here is an amazon link for a motor that should work.](https://www.amazon.com/gp/product/B00PNEPQ8E/ref=ppx_od_dt_b_slice_asin_title?ie=UTF8&psc=1)