bdring/midTbot_esp32 #12: Alternative to the 5V DC-DC Power Supply

bdring, 2019-08-29:
I have tested [this power supply from Alibaba](https://www.aliexpress.com/item/33034496914.html?spm=a2g0s.12269583.0.0.255713f1OnfaJm) and it works fine. It has a variable output rather than a fixed one, so it does need to be adjusted.

[image]

[image]

niekproductions, 2019-08-29:
I'm using [this fixed voltage one](https://nl.aliexpress.com/item/32950632281.html). It's a bit taller than a standard 7805, so you need to bend it forward a bit for the top cover to fit on. Otherwise it works fine!

[image]

dirkx, 2019-09-07:
These ones https://nl.aliexpress.com/item/32787145932.html work splendidly as well. They back of the board need to be facing 'outward'; the side with the pins inward.

admin-kepo, 2020-03-21:
can i replace it with IC L7805?

dirkx, 2020-03-21:
You can - but it will get very hot and will need a very sizable cooling fin.

aixlds, 2020-03-30:
Thank you for responding quickly to my question, If so I will try it with other modules that are similar because there are very few modules available in my place.

admin-kepo, 2020-03-30:
Thank you for replying, in the end I tried using a module that I found from aliexpress