bdring/midTbot_esp32 #13: Pen is lowered onto the paper after homing

niekproductions, 2019-08-29:
After completing the homing cycle, the bot proudly drops the pen down on the paper which leaves an unwanted dot in that spot.
Is there a way to disable this, and keep the pen up?

bdring, 2019-08-30:
I think there is a way to avoid that, but it will require some changes to the firmware. I'll look into that over the weekend.

bdring, 2019-08-31:
Fixed

I updated the firmware. The devt branch has the changes. Firmware changes are tested there before moving to the master branch.

The midTbot only homes the X and Y axis, but the servo axis feature of Grbl_ESP32 allows you to tell the servo what to do while the other axes are homing. It is setup to go to the max travel (z=5mm) while the other axes are homing. After homing is complete is goes to the current Z value. That value is set to a machine coordinate system value of 0 on all non homed axes. Z is a non homed axis. Therfore the Z was going to the down position after homing

You can have the servo use the machine coordinates or the work coordinates. Work coordinate systems allow you to do do things like set 0,0,0 points anywhere in the travel. Previously it was set to machine coordinates. I changed to use a work coordinate system. This allows you to offset the work coordinate system from the machine coordinate system by 5mm. This means when the machine system gets set to Z0 the work coordinate system is Z5, so the pen stays up.

You need to set the offset manually. You only need to do this once, because the machine will store it in flash memory. Send **G10L2P0Z-5** via a serial console or the console in in the WebUI.

You will need to compile and upload new firmware. Follow the [instructions](https://github.com/bdring/midTbot_esp32/wiki/Compiling-Firmware-for-the-MidTBot) here, but use the [Devt branch](https://github.com/bdring/Grbl_Esp32/tree/Devt).

I will update this post when it moves to the master branch

momo-aux, 2020-01-09:
Mine still drops the pen. Flashed this week with master. Is there a workaround or do I have to compile the dev Version

bdring, 2020-01-09:
I'll double check the master branch when I get home tomorrow.

bdring, 2020-01-10:
Just tested with latest master. It seems to be working fine for me. The CPU_MAP has a few features that control the behavior during homing.

```
#define SERVO_Z_HOMING_TYPE		SERVO_HOMING_TARGET // during homing it will instantly move to a target value
#define SERVO_Z_HOME_POS		SERVO_Z_RANGE_MAX
```

The servo should immediately go up when homing.

Are you sure your servo is going the right way? Try sending G0Z5 and G0Z0. Does it move up for Z5 and down for Z0? If not, send $3=4 to reverse the Z.

momo-aux, 2020-01-10:
Yes, its going up on G0Z5 and down on G0Z50
It is moving up before homing X but after Homing X it goes down.

This only happens when homing for the first time. 
After I raised the pen one time it stays up after homing.
But before homing the first time, I cant raise it. So it happens always after reconnecting the power and do homing for the first time.