bdring/midTbot_esp32 #14: No ESP32 chip listed in the BOM

dwightmulcahy, 2019-08-30:
Docs/mechanical_BOM.md does not list the ESP32 chip to get.

bdring, 2019-08-31:
Do not get the second one. It has the wrong number of pins.

It needs to be 19x2 with a pin width space of 0.9". Do not get the 15x2 version or the 19x2 1.0" width version.

dwightmulcahy, 2019-08-31:
Ugh, thanks... 

the specs on these seem hard to find but this one seems to meet the needs, 19x2 and ~0.89" pin spacing:
https://www.amazon.com/KeeYees-Development-Bluetooth-Microcontroller-ESP-WROOM-32/dp/B07QCP2451

bdring, 2019-08-31:
Most of my boards support the (2) widths of the 19x2 types.  I'll be adding this to the next rev of the PCB. I'll be ordering and testing it before I post it.

[image]

dwightmulcahy, 2019-09-03:
nice.  I bought one of the v1 boards that you had...

dwightmulcahy, 2019-09-04:
ended up getting this one: https://www.amazon.com/gp/product/B0718T232Z

closing this issue as the PCB/BOM.md lists the need for an ESP32 Dev Kit (0.9" Wide Type).

closed 2019-09-04 by dwightmulcahy