bdring/midTbot_esp32 #15: Unstable Y axis

niekproductions, 2019-08-31:
My midTbot is working great, but I've noticed that there is an increase in vertical play as the head gets further away from the X axis. It's up to around 5mm when fully extended, making the head drag across the paper. This also makes it more difficult to find a good pen height where it touches the paper across the full drawing surface.

Is this a common issue, or could I have bad bearings/rails?

bdring, 2019-08-31:
There should be no play in the Y axis. If there is a little play, it will exaggerate as the axis moves out. Try to find where the play is and fix it.

Not all printers have the same performance when creating hole sizes. If you have a rod or bearing that is loose in a hole, you may need to tweak to hole diameter. Many CAD programs, like fusion 360,  allow you to tweak STEP files. You also may be able to change some settings in your slicer.

niekproductions, 2019-08-31:
Thanks, I'll investigate where the play is. I suspect cheap bearings, but might have to tweak my parts as well.

I've already adjusted the hole diameter to have zero clearance, as the original parts had a very loose fit. Would you like me to post the new STLs somewhere?