bdring/midTbot_esp32 #18: plot file(s) with push button(s) ?

snug11, 2019-10-08:
hi all,
i`m asking for a possible (firmware) extension:
there are a few unused gpio`s on the esp32...could them be used to start different plot files by pushing buttons (internal pullup on/gnd) with ? 

for example:
gpio32/button1 = plot file "xxx.gcode"
gpio33/button2 = plot file "yyy.gcode"
(for best, pin assignment in ui under "sd-files area", beneath each loaded file)

that would be absolute easy to let guests (on a faire) start plotting different preloaded grahics without the needing of a second device (laptop, smartphone...)

...that would be great !  :-)

with kind regards from germany
andree

JamesCWhite, 2019-10-12:
See here:
https://github.com/bdring/Grbl_Esp32/wiki/Macro-Button-Feature

snug11, 2019-10-15:
I'll try it, thanks for showing that!  :-)