bdring/midTbot_esp32 #19: How to setup bluetooth?

BigDave365, 2020-02-01:
What is the best way to control the plotter via bluetooth (Windows 10)?

I assume a virtual com port, but don't know where to start on the windows side or the plotter side.

Thanks....

bdring, 2020-02-01:
The controller needs to have Bluetooth enabled. 

Make sure **#define ENABLE_BLUETOOTH** is not commented out in **config.h**
Give the Bluetooth a name with **[ESP140]MyBtName**
Put the radio in Bluetooth mode with **[ESP110]BT**

Connect from Windows.
Do an **Add a Bluetooth Device** from Windows.
It will create 2 new COM ports. Typically one is used by Windows to deal with the connection. The other can be used by you just like a regular COM port. Typically the higher number port is the one you use.

BigDave365, 2020-02-01:
Yeah that would have taken a while to sort out....

I do not understand the commands below:

Give the Bluetooth a name with [ESP140]MyBtName
Put the radio in Bluetooth mode with [ESP110]BT

Do I enter this on the command line on the terminal? 

This is my first foray into CAM (other than 3D printers), so sorry for all the questions.

Thanks for the help, I'll give it a go this afternoon.

bdring, 2020-02-01:
This GitHub repo is specific to the midtbot machine. These questions are more targeted at [Grbl_ESP32 firmware](https://github.com/bdring/Grbl_Esp32). You might find better resource on that repo.

The commands can be entered on any communications interface (USB,BT,Wifi), but USB is easiest because it is always on. It also gives a bunch of information when Grbl_ESP32 starts. This is before you could establish a wireless connection. It will tell you the Bluetooth or Wifi modes, addresses, etc.

The default USB/Serial settings are 115200,N,8,1. I would use the Arduino IDE serial monitor setup like below. (use your com port number of course)

[image]

You can click the reset button on your ESP32 to see all the boot information. All Grbl_ESP32 specific information is in this format [MSG: ....]

closed 2020-03-17 by bdring