bdring/midTbot_esp32 #21: GrabCAD interactive model not working

mahtDFR, 2020-03-17:
I navigated from here: https://github.com/bdring/midTbot_esp32#check-it-out-on-grabcad

It redirected me to this page:
https://workbench.grabcad.com/workbench/projects/gcj3zJAQexD3ve_8KkwymatyKXhCWnRs8TB5U1ojGxl3s4#/space/gcP-lh4vchvUQ6FbfQFYGKVWLmIdnV8aq2IyxzoECw8woR/link/1918044

I am met with the error message:
_This model does not appear to contain any geometry.
Failure reason: Unknown
Error reading model data from SCS file_

As a workaround I downloaded the STP file into Fusion360 which worked, but it would be nice to have it available in a browser for quick reference.

bdring, 2020-03-17:
The viewer appears to be be broken. I downloaded the file and was able to view it in my CAD program. It must be a GrabCAD problem. It might fix itself if the problem is widespread.

Leemonoid, 2020-06-26:
Just a quick follow-up to let you know the problem still exists. I can save the file and open my copy in the GrabCAD viewer, but the one you linked from your page will not open.

bdring, 2020-06-26:
[Here is a direct link to the viewer](https://workbench.grabcad.com/workbench/projects/gcj3zJAQexD3ve_8KkwymatyKXhCWnRs8TB5U1ojGxl3s4#/space/gcP-lh4vchvUQ6FbfQFYGKVWLmIdnV8aq2IyxzoECw8woR/link/1918044).

It seems to be stuck preparing the file. You could download the file and load in a CAD program. If you search for "online step file viewer", you should find some free options.