bdring/midTbot_esp32 #22: sdcard

vascobarbosa, 2020-03-25:
install it with this esp32 dev board 

https://cdn.shopify.com/s/files/1/1509/1638/files/ESP_-_32_NodeMCU_Developmentboard_Pinout_Diagram.jpg?4479111012146266271

but the sd card is not working, everything works, web ,through bluetooth, via telnet, via usb cable
but the sd_card is always not found. 

though about remapping the pins, but I guess I wanted to ask first if somebody had a clue

bdring, 2020-03-25:
Check the soldering of the SD card socket. We have seen a few poor connections on one of the batches. If you gently push each pin down with a soldering iron, that fixes it.

vascobarbosa, 2020-03-25:
thanks, gonna check them,

vascobarbosa, 2020-03-25:
Checked and rechecked the connections

with the multimeter everything is connected right, and on the right pins

on the we_gui I get always no SD card, and on the computer it works and format in fat32
on a mac osx

This is weird, and sorry about this, but this isnt one of your batchs, had some credits at jlcpcb.com and made them done some boards

bdring, 2020-03-25:
I would still suspect the soldering. Those pins can bend up during handling and not solder correctly.

Also see the [wiki about formatting the SD card](https://github.com/bdring/Grbl_Esp32/wiki/Using-the-SD-Card).

vascobarbosa, 2020-03-25:
fixed it, BAD SD FORMATING

On OSX open the terminal and follow this commands

#mount
/dev/disk2 on / (hfs, local, journaled)
devfs on /dev (devfs, local, nobrowse)
/dev/disk3s1 on /Volumes/USB_Disk (msdos, local, nodev, nosuid, noowners)
**In this case, my disk is disk3. Your disk will most likely be different! Be sure to reference the correct disk - you can and will lose your data if you format the wrong drive!!!**

3) Unmount the disk
#diskutil unmountDisk disk3
Unmount of all volumes on disk3 was successful

4) Format the drive
#sudo newfs_msdos -F 16 /dev/disk3
After a few moments, your drive will format.And it will work

closed 2020-03-26 by vascobarbosa