bdring/midTbot_esp32 #26: No Pen Lift in Z0 .. Z5 range

scruss, 2020-10-12:
Following instructions in [Pen Servo Setup](https://github.com/bdring/midTbot_esp32/wiki/Setting-Up-the-Motors#pen-servo-setup), servos (both MG90 and SG90 tried)  do not respond to `G0Z0` or `G0Z5`. `G0Z-180` results in a slow, juddering movement in the servo from 0 to ~90° clockwise. `$HZ` results in a prompt return to home, while `G0Z0` is slow.

## Boot messages

    [MSG:Grbl_ESP32 Ver 1.3a Date 20201004]
    [MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
    [MSG:Using machine:MIDTBOT]
    [MSG:Axis count 3]
    [MSG:RMT Steps]
    [MSG:Init Motors]
    [MSG:X  Axis Standard Stepper Step:GPIO(12) Dir:GPIO(26) Disable:None Limits(0.000,100.000)]
    [MSG:Y  Axis Standard Stepper Step:GPIO(14) Dir:GPIO(25) Disable:None Limits(-100.000,0.000)]
    [MSG:Z  Axis RC Servo Pin:27 Pulse Len(3276,6553) Limits(-100.000,0.000)]
    [MSG:Global stepper disable pin:GPIO(13)]
    [MSG:No spindle]
    [MSG:Client Started]
    [MSG:Connecting CB283W]
    [MSG:Connecting.]
    [MSG:Connecting..]
    [MSG:Connected with 192.168.2.134]
    [MSG:Start mDNS with hostname:http://grblesp.local/]
    [MSG:SSDP Started]
    [MSG:HTTP Started]
    [MSG:TELNET Started 23]
    [MSG:X  Axis limit switch on pin GPIO(2)]
    [MSG:Y  Axis limit switch on pin GPIO(4)]
    [MSG:'$H'|'$X' to unlock]

## Grbl_Esp32 settings

    $Sta/SSID=CB283W
    $Sta/Password=******
    $Sta/IPMode=DHCP
    $Sta/IP=0.0.0.0
    $Sta/Gateway=0.0.0.0
    $Sta/Netmask=0.0.0.0
    $AP/SSID=GRBL_ESP
    $AP/Password=******
    $AP/IP=192.168.0.1
    $AP/Channel=1
    $System/Hostname=grblesp
    $Http/Enable=ON
    $Http/Port=80
    $Telnet/Enable=ON
    $Telnet/Port=23
    $Radio/Mode=STA
    $Bluetooth/Name=btgrblesp
    $Notification/Type=NONE
    $Notification/T1=
    $Notification/T2=
    $Notification/TS=
    $Homing/Cycle5=
    $Homing/Cycle4=
    $Homing/Cycle3=
    $Homing/Cycle2=
    $Homing/Cycle1=X
    $Homing/Cycle0=Y
    $Report/StallGuard=
    $Spindle/Type=NONE
    $Stepper/Pulse=3
    $Stepper/IdleTime=255
    $Stepper/StepInvert=
    $Stepper/DirInvert=Y
    $Stepper/EnableInvert=Off
    $Limits/Invert=On
    $Probe/Invert=Off
    $Report/Status=1
    $GCode/JunctionDeviation=0.010
    $GCode/ArcTolerance=0.002
    $Report/Inches=Off
    $Limits/Soft=Off
    $Limits/Hard=Off
    $Homing/Enable=On
    $Homing/DirInvert=X
    $Homing/Squared=
    $Homing/Feed=200.000
    $Homing/Seek=1000.000
    $Homing/Debounce=250.000
    $Homing/Pulloff=3.000
    $GCode/MaxS=1000.000
    $GCode/MinS=0.000
    $GCode/LaserMode=Off
    $GCode/Line1=
    $GCode/Line0=
    $Spindle/Enable/Invert=Off
    $Spindle/Enable/OffWithSpeed=Off
    $Spindle/Delay/SpinDown=0.000
    $Spindle/Delay/SpinUp=0.000
    $Spindle/PWM/Invert=Off
    $Spindle/PWM/Frequency=5000.000
    $Spindle/PWM/Off=0.000
    $Spindle/PWM/Min=0.000
    $Spindle/PWM/Max=100.000
    $X/StepsPerMm=200.000
    $Y/StepsPerMm=100.000
    $Z/StepsPerMm=100.000
    $A/StepsPerMm=100.000
    $B/StepsPerMm=100.000
    $C/StepsPerMm=100.000
    $X/MaxRate=8000.000
    $Y/MaxRate=8000.000
    $Z/MaxRate=5000.000
    $A/MaxRate=1000.000
    $B/MaxRate=1000.000
    $C/MaxRate=1000.000
    $X/Acceleration=200.000
    $Y/Acceleration=200.000
    $Z/Acceleration=100.000
    $A/Acceleration=200.000
    $B/Acceleration=200.000
    $C/Acceleration=200.000
    $X/Home/Mpos=0.000
    $Y/Home/Mpos=0.000
    $Z/Home/Mpos=0.000
    $A/Home/Mpos=0.000
    $B/Home/Mpos=0.000
    $C/Home/Mpos=0.000
    $X/MaxTravel=100.000
    $Y/MaxTravel=100.000
    $Z/MaxTravel=100.000
    $A/MaxTravel=300.000
    $B/MaxTravel=300.000
    $C/MaxTravel=300.000
    $X/Current/Run=0.250
    $Y/Current/Run=0.250
    $Z/Current/Run=0.250
    $A/Current/Run=0.250
    $B/Current/Run=0.250
    $C/Current/Run=0.250
    $X/Current/Hold=0.125
    $Y/Current/Hold=0.125
    $Z/Current/Hold=0.125
    $A/Current/Hold=0.125
    $B/Current/Hold=0.125
    $C/Current/Hold=0.125
    $X/Microsteps=16
    $Y/Microsteps=16
    $Z/Microsteps=16
    $A/Microsteps=16
    $B/Microsteps=16
    $C/Microsteps=16
    $X/StallGuard=16
    $Y/StallGuard=16
    $Z/StallGuard=16
    $A/StallGuard=16
    $B/StallGuard=16
    $C/StallGuard=16

bdring, 2020-10-12:
I think you just need to optimize the range, homing mpos homing direction.

https://github.com/bdring/Grbl_Esp32/wiki/Machine-Space-and-Homing
https://github.com/bdring/Grbl_Esp32/wiki/Servo-Axis-Feature

AndyRockall, 2021-03-17:
I'm having the same problem, has anyone solved this yet.
I'm using the same electronic/mechanical setup as in Bart's setup and build instructions.
I'm using Vers 2 of the board and the new pen lift, everything works except the pen lift servo.
I've checked, double checked and triple checked everything and am starting to tear my hair out (what's left of it).

Bart,
Have you tried your MidTbot with the new version of GRBL_ESP32?
If you have and it works could you perhaps publish your settings somewhere?

bdring, 2021-03-17:
I updated mine and used it within the last week. Have you read this page?

https://github.com/bdring/Grbl_Esp32/wiki/Servo-Axis-Feature

Is the Z not moving at all?

Paste your boot messages here. I can look at them.

https://github.com/bdring/Grbl_Esp32/wiki/Requesting-Help#boot-messages

AndyRockall, 2021-03-18:
No the Z isn't moving at at all.
boot messages below

[MSG:Grbl_ESP32 Ver 1.3a Date 20210226]
[MSG:Compiled with ESP32 SDK:v3.3.4-432-g7a85334d8]
[MSG:Using machine:midTbot]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:Global stepper disable pin:GPIO(13)]
[MSG:X  Axis Standard Stepper Step:GPIO(12) Dir:GPIO(26) Disable:None Limits(5.000,205.000)]
[MSG:Y  Axis Standard Stepper Step:GPIO(14) Dir:GPIO(25) Disable:None Limits(-150.000,0.000)]
[MSG:Z  Axis RC Servo Pin:27 Pulse Len(3276,6553) Limits(5.000,10.000)]
[MSG:Servo Update Task Started]
[MSG:CoreXY (midTbot) Kinematics Init]
[MSG:No spindle]

[MSG:Client Started]
[MSG:Connecting sueandandy]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connecting....]
[MSG:Connected with 192.168.0.170]
[MSG:Connected with 192.168.0.170]
[MSG:Start mDNS with hostname:http://MidTbot.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:X  Axis limit switch on pin GPIO(2)]
[MSG:Y  Axis limit switch on pin GPIO(4)]

bdring, 2021-03-18:
Send $Z and paste the response here.

AndyRockall, 2021-03-18:
$Z/RcServo/Cal/Max=1.000
$Z/RcServo/Cal/Min=1.000
$Z/StepsPerMm=100.000
$Z/MaxRate=5000.000
$Z/Acceleration=100.000
$Z/Home/Mpos=5.000
$Z/MaxTravel=5.000
$Z/Current/Run=0.250
$Z/Current/Hold=0.125
$Z/Microsteps=16
$Z/StallGuard=16
ok

bdring, 2021-03-18:
Is the servo locked or can you easily turn it by hand?

Also, send $Homing and paste the response.

AndyRockall, 2021-03-18:
The servo doesn't seem to be locked, I also measured the voltages and have 5V supply and about 0.04V on the signal pin.

$Homing/Cycle0=Z
$Homing/Cycle1=Y
$Homing/Cycle2=X
$Homing/Cycle3=
$Homing/Cycle4=
$Homing/Cycle5=
$Homing/Enable=On
$Homing/DirInvert=XZ
$Homing/Squared=
$Homing/Feed=500.000
$Homing/Seek=2000.000
$Homing/Debounce=250.000
$Homing/Pulloff=3.000

bdring, 2021-03-18:
Try sending $Homing/DirInvert=X

Then reset the ESP32 and show me the boot text.

I would like to see this 
[MSG:Z Axis RC Servo Pin:27 Pulse Len(3276,6553) Limits(5.000,10.000)]
change to this 
[MSG:Z Axis RC Servo Pin:27 Pulse Len(3276,6553) Limits(0.000,5.000)]

AndyRockall, 2021-03-18:
Looks like you got what you expected.

MSG:Grbl_ESP32 Ver 1.3a Date 20210226]
[MSG:Compiled with ESP32 SDK:v3.3.4-432-g7a85334d8]
[MSG:Using machine:midTbot]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:Global stepper disable pin:GPIO(13)]
[MSG:X  Axis Standard Stepper Step:GPIO(12) Dir:GPIO(26) Disable:None Limits(5.000,205.000)]
[MSG:Y  Axis Standard Stepper Step:GPIO(14) Dir:GPIO(25) Disable:None Limits(-150.000,0.000)]
[MSG:Z  Axis RC Servo Pin:27 Pulse Len(3276,6553) Limits(0.000,5.000)]
[MSG:Servo Update Task Started]
[MSG:CoreXY (midTbot) Kinematics Init]
[MSG:No spindle]

AndyRockall, 2021-03-18:
Bart,
Just remembered I had to increase the X/Y steps per mm to 200 to get it to travel the correct distance, does this shed any light?

bdring, 2021-03-18:
Here is an invitation to our discord server. It might be easier to help you there.

There is a midtbot channel.

https://discord.gg/kFJTzHFN4y

AndyRockall, 2021-03-18:
Bart,
I decided to re-compile with VSC rather than Arduino IDE.
Hey Presto it now works!
Compiled with Arduino IDE again and it didn't work.

So it seems as if we have to compile with VSC for it to be successful..

Thanks for all your help and the late night (it was 4:30 AM when I went to bed), much to my wife's displeasure.

odaki, 2021-06-07:
This change may improve the situation.
https://github.com/bdring/Grbl_Esp32/pull/914