bdring/midTbot_esp32 #27: Limit switches

dunkfordyce, 2021-01-06:
Hi,

I have used the files here to print and make my first plotter and in fact first use of grbl and the first thing I have 3d printed that is in some way useful!

Due to the PCB not being available and because I'm just learning I have put together the electronics on a bread board. Amazingly I have managed to get it drawing. Thank you for the excellent project. 

I am now trying to get the limit switches working. In the schematic files in this repo the switches look like they are normally open, but the internet suggests that limit switches for this sort of application are normally closed and made open when the limit is hit. So should they be normally open or closed?

Thanks

bdring, 2021-01-07:
You can use either type. They can be inverted in the software by sending $Limits/Invert=On or Off

https://github.com/bdring/Grbl_Esp32/wiki/Settings#limitsinvert-or-5---limit-pins-invert-boolean

dunkfordyce, 2021-01-07:
Ah thankyou, I tried that but I didnt have any pull down resistors so I'm
not sure if it was ever going to work. I ended up changing my switches to
be normally closed as this document recommends
https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches.

If I reset all my settings with `$rst=$` (so using defaults from midtbot.h)
and then do a $H the y axis correctly goes towards the center of the bot
and will stop at the limit, but after that, when it does the X homing, it
goes in the wrong direction (towards the right/positive X).

dunkfordyce, 2021-01-07:
One thing I have never understood with the instructions in this repo is which way around and where 0,0 should be! 

I ended up with the pen side closest to me, and then moved the pen to the bottom left as my 0, 0. So that positive y goes away from me and positive X goes to my right. Is that correct? because if its not then all my questions are back to front and sideways.

bdring, 2021-01-08:
It does not matter where the machine 0,0 is, but positive Y should move towards the center and positive X should move to the right.

You can then manually jog to where you want the 0,0 and set it.